adafruit/Adafruit_VEML7700 #1: Lux calc fix

ladyada, 2019-05-13:
https://forums.adafruit.com/viewtopic.php?f=19&t=149893&p=748684#p748675

siddacious, 2019-05-14:
@ladyada OP says the lux correction is to be applied only at 25ms 1/8 gain

ladyada, 2019-05-14:
hmm unclear. i think it should be a new function called lux_normalize() that will turn the plain lux thru the special function

ladyada, 2019-07-04:
https://forums.adafruit.com/viewtopic.php?f=19&t=152997&p=758582#p758582

jeff-winn, 2021-10-30:
I was recently rewriting the driver for this in C# for use in a .NET Core application and may have found the crux of the issue.

On the data sheet for the VEML7700, the document clearly contains a multiplier table based on the values of gain and integration time.

This driver has hard coded a single value out of that table rather than properly adjust the multiplier. Your multiplier is hard coded at 0.0576! This likely means all numbers produced other than those whose settings match are off. 

The methods in this driver would be normalize_resolution and readLux where that number has been hard coded. This is likely why there is other logic to try and readjust the value output.

Here’s a direct link to the PDF I kept a copy of so I wouldn’t have to track it down again…
https://github.com/jeff-winn/dotnet-veml7700/blob/main/docs/Adafruit_VEML7700/designingveml7700.pdf

The multiplier table is on page 5 of that document.

Here’s a link to what I wrote, the AdjustLuxMultiplier method in the class is the important one…
https://github.com/jeff-winn/dotnet-veml7700/blob/main/src/Adafruit.Devices.Veml7700/Adafruit_VEML7700.cs

caternuson, 2022-05-16:
Yah, sure seems like it:
[image]

caternuson, 2022-05-26:
Hopefully fixed with the 2.0.0 release:
https://github.com/adafruit/Adafruit_VEML7700/releases/tag/2.0.0

closed 2022-05-26 by caternuson