adafruit/Adafruit_VEML7700 #12: "Adafruit Unified Sensor" missing from the 'depends' statement in 'library.properties' file

glyndon, 2021-06-30:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**
ESP8266

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**
N/A, using platformio

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

During compile, errors result from omission of the "Adafruit Unified Sensor" library, because it is missing from the 'depends' statement in this repo's 'library.properties' file

ladyada, 2021-06-30:
it must be some other part of your code. 
https://github.com/adafruit/Adafruit_VEML7700/search?q=Adafruit_Sensor

closed 2021-06-30 by ladyada

glyndon, 2021-06-30:
The "Adafruit Unified Sensor" library *is* included on the depends line in the BMP3XX library's properties file. So the behaviour between the two is not consistent, hence surprising and thought you might want them to be consistent.
I was able to workaround this by naming the needed library as a dependency in my ESPHome YAML file, even though it wasn't needed for another project that uses the BMP3XX library.

ladyada, 2021-06-30:
yep - but this isnt the BMP3xx library - it does not have that dep :)

glyndon, 2021-06-30:
Wouldn't compile until I added it.
All I'm trying to bring up is that it may not be needed by Arduino's preprocessor, but that it does seem to be needed for ESPHome's compiler.  In the separate project (the one with BMP3xx), that lib wasn't mentioned either, but got included by 'depends', so everything has always compiled fine.
In the VEML's library it's not in 'depends', not mentioned in the docs, and an ESPHome compile fails without it.

ladyada, 2021-06-30:
sounds like ESPHome needs to add it then, it isnt a dep for this librayr. please tell ESPHome to fix

glyndon, 2021-06-30:
You are correct. Mine was a sloppy workaround - adding the Unified lib must've caused another inclusion that pio needed. 
Adjusting the YAML to specify only the 7700 and BusIO libs takes care of it nicely. I'll pass this along to ESPHome if they ever compose a built-in integration for it. I had to build this one as a 'Custom Sensor' by hand, as I had done for the BMP388.
Thanks for the needed details to make this build cleanly.