adafruit/Adafruit_VEML7700 #13: Bug in example code

mdede439, 2022-05-15:
I could not get the veml7700_test example code to work.  I found the first few lines of the Setup as follows:

void setup() {
  while (!Serial) { delay(10); }
  Serial.begin(115200);
  Serial.println("Adafruit VEML7700 Test");
  ......

When I run the sketch the serial monitor shows nothing.  I see that the while(!Serial) comes before the Serial.begin so as far as I understand it the while statement is looking for a serial input before Serial is initialized.  I can't see this ever working so I swapped those lines and everything works.

caternuson, 2022-05-15:
Thanks. Yah, that's probably better to be swapped. Want to PR that change?

mdede439, 2022-05-15:
PR that change? I don't understand.

caternuson, 2022-05-15:
PR = pull request. It's how you could submit those code changes to the example. But it's not required. Just suggesting to see if you wanted to submit.

mdede439, 2022-05-16:
Oh yes. I get it.  I am not a big user of GitHub so wouldn't know exactly
how to do that.

Michael D

closed 2022-05-21 by ladyada