adafruit/Adafruit_VEML7700 #18: Auto lux not working for low light levels

caternuson, 2022-06-24:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=192254

Recreated locally.
[image]

caternuson, 2022-06-28:
Closing. Hopefully fixed with #19.

closed 2022-06-28 by caternuson

bmtully, 2023-01-30:
@caternuson might be related, the autoLux function seems to get stuck when lux drops to zero

If you cover the sensor, then run readLux(VEML_LUX_AUTO), the function won't return until the sensor is uncovered...

caternuson, 2023-01-30:
@bmtully Please open a new issue for this. Include what sketch you are running and what the Serial Monitor output looks like.