adafruit/Adafruit_VEML7700 #23: Auto Lux gets stuck in dark conditions

bmtully, 2023-02-04:
- Arduino board:  Adafruit QT Py ESP32-S2

- Arduino IDE version (found in Arduino -> About Arduino menu):  2.0.3, running ESPHome

- List the steps to reproduce the problem below:

In this code I am trying to poll a VEML7700 every 15 seconds, to send back the "auto lux" value. 

This works well, until the sensor is no longer exposed to any light. Once the light level drops to 0, the auto lux function will not return anything until the sensor is re-exposed to light

For example, if you cover the sensor with a cup, the auto lux function will start, but it will not return a value until you pick up the cup. In my case this blocks the entire system and the ESP device shuts down... until you pick up the cup.

The cup is just an example, I noticed this when all of my sensors started going offline at night

```
#include "esphome.h"
#include "Adafruit_VEML7700.h"

class VEML7700 : public PollingComponent, public Sensor {
    public:
    Adafruit_VEML7700 veml;
    
    VEML7700() : PollingComponent(15000) {}

    float get_setup_priority() const override { return esphome::setup_priority::DATA; }

    void setup() override {
        Serial.begin(115200);
        while (!Serial) { delay(10); }
        Serial.println("Adafruit VEML7700 Auto Lux Test");

        if (!veml.begin()) {
            Serial.println("Sensor not found");
            while (1);
        }
        Serial.println("Sensor found");
    }

    void update() override {

 // When light level drops to 0, Serial will print the first statement,
// but the second will not come until the sensor is re-exposed to light.
// If I remove "VEML_LUX_AUTO", the value is always returned immediately

        Serial.println("Calculating Auto Lux...");
        float lux = veml.readLux(VEML_LUX_AUTO);
        Serial.print("Lux = "); Serial.println(lux);

        publish_state(lux);
        
    }

};
```

Here is the serial output, with notes added when the sensor was covered and re-exposed to light:

```
[D][api:102]: Accepted ::FFFF:10.0.1.3
Calculating Auto Lux...
Lux = 5.70
[D][sensor:127]: 'VEML7700': Sending state 5.70240 lux with 0 decimals of accuracy
Calculating Auto Lux...
Lux = 5.82
[D][sensor:127]: 'VEML7700': Sending state 5.81760 lux with 0 decimals of accuracy
**  Sensor Covered
Calculating Auto Lux...
Lux = 0.00
[D][sensor:127]: 'VEML7700': Sending state 0.00000 lux with 0 decimals of accuracy
Calculating Auto Lux...
[Disconnected]
**   Wait 5 Minutes
**   Uncover Sensor
[Connected]
Sensor found
[C][wifi:037]: Setting up WiFi...
Calculating Auto Lux...
[Disconnected]
[Connected]
Adafruit VEML7700 Auto Lux Test
Sensor found
[C][wifi:037]: Setting up WiFi...
Calculating Auto Lux...
Lux = 5.47
[D][sensor:127]: 'VEML7700': Sending state 5.47200 lux with 0 decimals of accuracy
```

caternuson, 2023-02-04:
Is this behavior unique to ESPHome? Do you get the same behavior running the example from the library?
https://github.com/adafruit/Adafruit_VEML7700/blob/master/examples/veml7700_autolux/veml7700_autolux.ino

bmtully, 2023-02-04:
Using that exact sketch with Arduino IDE -- yes, everything works as expected! So this might be unique to ESPHome. 

It seems like the process of looping through the gain/integration times is taking longer than expected and the ESP shuts down. 

What do you think, is this something that should be resolved on the ESPHome side? If the ESP forces itself to reboot mid-process then it makes sense why the sensor becomes unresponsive. But on the other hand why is the auto lux function taking so long on the 2nd zero reading? When the sensor is covered, the next reading will be 0 lux -- it loops through the gain/integration times, settles on the highest ones, and returns 0 lux. Perfect. But the following reading, where it starts at 0 lux, will never complete that gain/integration loop, so the watchdog is timing out. 

Total amateur here-- apologies if I'm posting in the wrong place-- just trying to understand the best I can

caternuson, 2023-02-05:
Not familiar with ESPHome and it's general requirements for library code. Based on the code snippet in your first post, it looks like it does not use the library simply and directly. Instead it uses a newly created class `VEML7700` which then calls `veml.readLux(VEML_LUX_AUTO)` inside its `update()` method.

The auto lux reading will end up making a blocking delay call that will vary depending on current integration time:
https://github.com/adafruit/Adafruit_VEML7700/blob/5c4a5a50236c202dd7edafdb2b4913f754845261/Adafruit_VEML7700.cpp#L389

ESPHome seems to not be happy with that. The long integration times used for low light end up causing a WDT reset.

So this may be an incompatibility issue - this library won't work as is with ESPHome.

caternuson, 2023-05-02:
Closing. Issue is usage in ESPHome.

closed 2023-05-02 by caternuson